peterhinch/micropython_eeprom #3: Unable to implement this library on ESP32 WROVER-B

lnsriravichandran22, 2020-04-09:
Dear Pete!!

I hope that you are safe. 

I am getting this message when I try to run 
[image]

FYI, I am using ESP32-Azure IoT Kit with ESP32-WROVER-B.

Please advise !!

peterhinch, 2020-04-10:
Please could you explain what Flash chips you are using and how they are wired to the ESP32 chip.

The `wemos_flash.py` script is really intended for ESP8266 and I've only run it on that target. However the library should work on ESP32 if you are using supported chips and they are wired correctly.

lnsriravichandran22, 2020-04-10:
Pete, As per the BOM , the flash part number is 'U3'. I could not find any reference to the part number of the flash memory anywhere!!

From the schematic of ESP32-WROVER-B , all I see the below pin mapping as follows :

[image]
[image]

The pin connections are as above.

Please advise!!

peterhinch, 2020-04-10:
It's difficult for me to support an unknown chip. I'd intended these drivers to be used by people building their own hardware with one of the supported chips which I bought and tested. There is some standardisation of Flash devices in the form of a JEDEC spec, but yours may have some characteristic which is causing this behaviour. Even if we knew the chip type I'd have to buy some to investigate.

Assuming you've defined the pins correctly in the code the only thing I can think to try is to reduce the baudrate, say from 20MHz to 5MHz.

MugenKadai, 2020-07-31:
Pete, I am extremely sorry.

This is a complete chaos and I assumed that these are intended for the internal flash memory that comes with ESP.

Thanks again for the work!!

lnsriravichandran22, 2020-07-31:
Pete,

The mistake is all mine and hence the chaos.

I thought this is intended for internal Flash. 

Closing this. Sorry again for wasting your time!!

closed 2020-07-31 by lnsriravichandran22