peterhinch/micropython_eeprom #4: Winbond Flash: Wrong setting for a WinBond W25Q64FV device

robert-hh, 2020-06-07:
Hello Peter, thank you very much for the excellent drivers. I tried that this afternoon with a WinBond w25q64fv chip (8 MByte). It did not work out-of-the box. When comparing the data sheet with the code if saw, that it requires command set you chose for devices <= 4 MB. With changíng that decision to <= 8 MByte the driver works. It might be only this chip which requires the change. But in case someone asks, here is a piece of information.

peterhinch, 2020-06-08:
Thanks for that. Perhaps I should add an optional constructor arg `cmd_len=None`. This would enable the default to be overridden by specifying 4 or 5 byte commands.

robert-hh, 2020-06-08:
That sounds like a good idea, since not only the length of the commands change, but also the commands itself change according to the command length. e.g.:
4 byte command: read=0x3
5 byte command: read=0x13
And it does not depend on the chips size alone.
Windbond 16MByte flash: 4 byte command
Cypress 16 MByte flash: 5 byte command by default, can be changed to 4 byte.

peterhinch, 2020-06-08:
Now done. Please let me know the outcome with your chips.

robert-hh, 2020-06-08:
Thanks for the swift fix. It works. Only the options False/True for a keyword 'cmdset' is a little bit irritating. So maybe y keyword like 'longcmd' or 'cmd4' would better match a False/True option, or a keyword 'cmdlen' with the options 4/5.