peterhinch/micropython_eeprom #5: can't able to perform read/write operation

Ajaynithish, 2020-10-19:
Hi
I am using S25FL128S Flash NOR memory from cypress it's size is 256Mb(32MB). I have connected it with G01(an OEM module from pycom). when I try to run little_fs.py it shows error at fcreate() function call(OSERROR). It can't able to mount the memory.
Then I tried flash_test.py I was able to write single byte using`flash[2000]=42` snippet you mentioned but can't able to use second snippet writing a byte array`flash[2000:2002] = bytearray((42, 43))`. For that I am getting the following error
`File "main.py", line 7, in <module>
  File "/flash/lib/bdevice.py", line 26, in __setitem__
  File "/flash/lib/bdevice.py", line 46, in _wslice
  File "/flash/lib/bdevice.py", line 37, in _do_slice
AttributeError: 'slice' object has no attribute 'step'` and I too can't find how those step,start,stop attributes came there in _do_slice function.
And I am not clear about why two chip select pins(does it for two external flash memory in single spi?).As I am using single chip I have done it this way will it cause any issue?
`spi = SPI(0, mode=SPI.MASTER, baudrate=10000000, polarity=0, phase=0, pins=('P10','P11','P14'))#clk,mosi,miso
cspins = (Pin('P8', Pin.OUT, value=1),) 

flash = FLASH(spi, cspins)`
Could you please explain am I doing it the right way or making any logical mistakes @peterhinch ?
could you please share your private contact details?

peterhinch, 2020-10-19:
You have a firmware issue. What version are you running?
> I too can't find how those step,start,stop attributes came there in _do_slice 

These come from the [Python slice class](https://docs.python.org/3/library/functions.html#slice). The latest MicroPython firmware supports this, but I'm guessing that the version you're using does not. I suggest you install the latest daily build, or the current release build V1.13. I trust you are running official MicroPython: if you're running a fork all bets are off.
> And I am not clear about why two chip select pins

My testing was done on a pair of chips to verify that the pair were correctly configured as a single array, hence the two pins - you need one pin for each chip. With a single chip your code looks correct.
```python
cspins = (Pin('P8', Pin.OUT, value=1),)
```
The chip you have uses a different technology from the S25FL128L I tested. I'm unsure about this in S25FL128S datasheet section 1.2.2.4:
> The command for erasing a 4-KB sector is supported only in the 128-Mb and 256-Mb density FL-S devices and only for use on the thirty two 4-KB parameter sectors at the top or bottom of the device address space.

The driver uses 4KB blocks throughout, and will only work with chips that fully support that block size.

It is not really practical for me to do a detailed analysis of chips I don't own and can't test. I'm grateful for positive feedback from users who have had success with other chips and will report it, but that's as far as I can go. Your firmware problem needs to be fixed to enable testing to proceed.
> could you please share your private contact details?

I prefer to do support in a public forum as it constitutes a resource which can help other users.

Ajaynithish, 2020-10-20:
> You have a firmware issue. What version are you running?
>
>
> These come from the [Python slice class](https://docs.python.org/3/library/functions.html#slice). The latest MicroPython firmware supports [...]

Ajaynithish, 2020-10-20:
I am using a pycom board and pymakr package recommended for it in VScode. it shows this 
`Pycom MicroPython 1.20.1.r1 [v1.11-3138a13] on 2019-10-08; GPy with ESP32`
`Pybytes Version: 1.1.3`

peterhinch, 2020-10-20:
This evidently doesn't support the Python slice class (and may have other issues). My driver requires official MicroPython. Your options are:
 1. Use official MicroPython. This may not support features specific to your Pycom hardware.
 2. Adapt my driver to match your firmware. May be tricky depending on how broken the Pycom firmware is.
 3. Look for another driver which uses only basic Python functionality
 4. Try to get Pycom to improve their Python support. It's possible that features such as slice support have been removed deliberately to conserve some resource (RAM, Flash).

In all cases you'll need to assess the chip datasheet, then test whether your chip works.

I'm not prepared to support forks of MicroPython. The workload is too much, and in this case would mean discarding valuable functionality. Sorry I can't offer more.

closed 2020-10-20 by peterhinch