peterhinch/micropython_eeprom #6: Why is supported chip size floored at 64Kbits?

jphalip, 2021-04-22:
Hi, is there a reason why the list of available chip sizes is floored at 64Kbits? Wouldn't it work with smaller sizes?

peterhinch, 2021-04-23:
I guess you're talking about EEPROMs, but the comment below is true for all types.

The list of supported chips represents chips tested by myself or others. It is very possible that other chips will work as there is a good deal of standardisation, but the only way to be sure is to check the datasheet of the chip you intend to use. If you are successful in using another chip, please report it and I will update the docs.

jphalip, 2021-04-23:
Yes sorry, I forgot to mention it was about EEPROMS. I've successfully tested the library with a CAT24C256LI-G chip. I currently do not have smaller chips to test with, although I might in the future. I was mainly curious if there was a technical limitation, as the library currently raises an exception if the provided chip size doesn't match one of the existing constants.

Maybe instead of raising an exception, it could simply display a warning?

peterhinch, 2021-04-24:
Yes, I guess I was being overly restrictive. There is more standardisation between chips than I realised when I started out. I'll make this change in due course. Thanks for testing and reporting back.

peterhinch, 2021-04-25:
I have made this change for I2C and SPI EEPROMs, also updated the README to report your success with CAT24C256LI-G. 

Please report success with other devices by an issue or a README.md PR.

closed 2021-04-25 by peterhinch