peterhinch/micropython_eeprom #7: when the dictionary is too large, write data incorrectly

QQWE1345, 2021-05-08:
[image]

I encountered this problem in use, when the dictionary is too large, write data incorrectly

QQWE1345, 2021-05-08:
below is the code:

`d = {0: '', 1: '{G0001#000P0500T9000!#001P0996T0995!}', 2: '{G0002#000P0500T9000!#001P0996T0995!}', \
3: '{G0002#000P0500T9000!#001P0996T0995!}', 4: '{G0002#000P0500T9000!#001P0996T0995!}', 5: '{G0002#000P0500T9000!#001P0996T0995!}'}
wdata = ujson.dumps(d).encode('utf8')
sl = '{:10d}'.format(len(wdata)).encode('utf8')
eep[0 : len(sl)] = sl  # Save data length in locations 0-9

print(eep[:10])

start = 128  # Data goes in 10:
end = start + len(wdata)
print(start," ",end)
eep[start : end] = wdata
print(wdata)
print(eep[start : end])
#pyb.delay(300)

#slenx = int(eep[:10].decode().strip())  # retrieve object size
#startx = 128
#endx = startx + slenx
#dx = ujson.loads(eep[startx : endx])
#print(dx)`

peterhinch, 2021-05-08:
You are running very old firmware - the first thing I would do is update to V1.15 and repeat.

Have you run the tests in the [test program](https://github.com/peterhinch/micropython_eeprom/blob/master/eeprom/i2c/I2C.md#5-test-program-eep_i2cpy) to check your hardware? You may have no need of the filesystem test but I would definitely run full_test as it is thorough and would definitely expose an error of the type seen. What EEPROM chip are you using?

It seems to be behaving as if the data were written at a start location of 64 rather than 128, so that when you display data at [128:end] you are missing the first 64 bytes. It would be worth checking if the data at [64:129] matches this theory. I can't explain this at all, given that the same method is used for reading and writing.

QQWE1345, 2021-05-10:
[image]

with V1.15 I get the result above,

my chip is  AT24C256

peterhinch, 2021-05-10:
The 128 byte page size is hard coded in the driver. While you could change this, it might be easier just to buy one of the supported chips. Good luck with whatever you decide to do.

closed 2021-05-10 by peterhinch

QQWE1345, 2021-05-10:
with my chip ,  just  change 128 to 64  in  full_test()   and everyyhing will be ok