peterhinch/micropython_eeprom #10: Example project specifically for the 25LC512

ph1lj-6321, 2022-02-11:
Hi
Any chance of a complete project example - I have reviewed and attempted to implement something on the Tiny2040, alas getting no where fast

peterhinch, 2022-02-11:
I'm not sure what you mean by a complete project. The [test scripts](https://github.com/peterhinch/micropython_eeprom/blob/master/eeprom/spi/eep_spi.py) are what I used in developing these drivers.

The 25LC512 is not one of the chips I tested, but given that the 25xx1024 works there should be no problem if you adapt [the instructions](https://github.com/peterhinch/micropython_eeprom/blob/master/eeprom/spi/SPI.md), notably the chip size constructor arg.

ph1lj-6321, 2022-02-11:
[image]

This is what is returned - note the bdevice & eeprom_spi are on the device

ph1lj-6321, 2022-02-12:
Thank you, that was a silly mistake on my part - thanks for your prompt response

[image]

After running the Write_Read

Not that it matters - but I get nothing returned when I run the "eep_spi.py"

Thanks again

peterhinch, 2022-02-12:
> Not that it matters - but I get nothing returned when I run the "eep_spi.py"

Please read [the docs](https://github.com/peterhinch/micropython_eeprom/blob/master/eeprom/spi/SPI.md#5-test-program-eep_spipy). Nothing will happen on import but the tests listed in the doc can be run.

```python
>>> import eep_spi
>>> eep_spi.test()
```
etc. If it runs `full_test` the chip can be categorised as working :)

peterhinch, 2022-02-18:
Did the chip pass the tests?

ph1lj-6321, 2022-02-18:
Hi sorry

Yeap - thanks for your help

peterhinch, 2022-02-19:
Thanks for the feedback. I've updated `README.md`, removed the driver's warning message for that chip size, and added some text from `eep_spi.py` to explain usage when it's initially imported.

closed 2022-02-19 by peterhinch