peterhinch/micropython_eeprom #11: T24C64 (using 24LC64 I2C) page write buffer is only 32 bytes

pretorhv, 2022-06-09:
Hi,

Newbie - treat with suspicion. 

I am using eeprom_i2c to write a JSON encoded config (similar to your example), and it seems to write only 32 bytes out of every 128 using either slicing or readwrite. The 24LC64 datasheet states the pagewrite buffer is 32 bytes.

Henk

peterhinch, 2022-06-10:
The first step is to run [the test script](https://github.com/peterhinch/micropython_eeprom/blob/master/eeprom/i2c/eep_i2c.py). You'll need to change [this line](https://github.com/peterhinch/micropython_eeprom/blob/295a10396508de0bb1e0cf67327bea2e2a7eeea9/eeprom/i2c/eep_i2c.py#L9) to match your chip and [this line](https://github.com/peterhinch/micropython_eeprom/blob/295a10396508de0bb1e0cf67327bea2e2a7eeea9/eeprom/i2c/eep_i2c.py#L17) for chip and I2C number. The first test to run is `test()`. Then try the filesystem based tests.

Your chip is a supported device so these tests have been run successfully.

peterhinch, 2024-01-09:
There was a bug which was probably causing your problem. Please try the latest version.