peterhinch/micropython_eeprom #12: problem with `W25Q128` on rp2040

r00t1900, 2022-06-20:
I follow the `Flash.md` and copy `flash_spi.py` `bdevice.py` to my project. My chip is `W25Q128FV` with a size of 16MiB.

Here is my code:
```python
import os
from machine import SPI, Pin
from lib.flash_spi import FLASH

cspins = (Pin(17, mode=Pin.OUT, value=1),)
flash = FLASH(SPI(0, baudrate=20_000_000), cspins, cmd5=True) 
```

After running this I got this output: **1 chips detected. Total flash size 55213970....394368MiB**. That is really not the size of my flash.

Besides, when I use this method to set flash value:
```python
flash[2000]=12
print(flash[2000])
```
I can got the correct result before reboot. But after reboot and read value from flash again, the value fallback to 255(`0xff`), so is this a false writing?

peterhinch, 2022-06-21:
To answer your second question, see [the docs](https://github.com/peterhinch/micropython_eeprom/blob/master/flash/FLASH.md#sync).

But clearly something is wrong. This chip was tested by a user and I don't have a sample, so any comment from me is something of a guess. Are you sure the `cmd5` [constructor arg](https://github.com/peterhinch/micropython_eeprom/blob/master/flash/FLASH.md#411-constructor) is correct?. The user (in the docs) says it should be `False`.

r00t1900, 2022-06-21:
Because I use a 128Mb chip so I guess I need to use cmd5=True to support bigger addresses searching. Am I wrong?>  The user who test the chip is a W25Q32chip which has only 32Mb.

peterhinch, 2022-06-21:
I quote from the docs:

>Winbond W25Q128JV Flash 128MiB tested by https://github.com/peterhinch/micropython_eeprom/issues/8#issuecomment-917603913.
This requires setting cmd5=False.

r00t1900, 2022-06-24:
It's so weird...no matter the cmd5 is set to True or False, the chip size return in a very very large number...The number is 55213970774324510299478046898216203619608871777363092441300193790394368MiB.Is it overflow???

r00t1900, 2022-06-24:
Oh I find out why. The reason is the inappropriate spi pin definitions.I use the rp2040 official pin layout, which indicates that #21 - #25 is the default SPI0. However in micropython documentation v1.18 there are no default spi0 definition that is same as rp2040 official. That is so misleading and I am so lucky to find out this.By the way, the micropython-stubs v1.18 did not has a clear description of machine.SPI class and this also lead to this problem. I hope we will have a better micropython-stubs under vscode or any other IDE.in conclusion, the usage of:```pythonfrom machine import SPIspi=SPI(0, 20_000_000)```did not point out what is the  default SPI0 pin layout and the raspberry Pi official documentation just point out a different of Out-of-date layout.Now I connect my external SPI flash on #8, #9, #10 and #11 which is a Spin :)

r00t1900, 2022-06-24:
Even the erase() won't work...

r00t1900, 2022-06-24:
Here is my write part code in REPL:```pythonflash[0]=123print(flash[0])# return 123flash.sync()# return 0```Then I reboot, and view data in REPL:```pythonflash[0]# return 0```

r00t1900, 2022-06-24:
Finally, I make it. The cmd5 really need to be set into False so the `flash.sync()` could be executed correctly!## Conclusion- Make sure the `machine.SPI` is initialize with the correct Pin layout.- For Winbond Series Flash chip, the argument `cmd5` need to be a False.- To make byte level write to take effect, always remember to call a `flash.sync()` so that the change could be applied from RAM to Flash.:) So glad that I finally fix it. Thank you Peter :)

peterhinch, 2022-06-24:
I'm glad you've got it working. I was just about to come back to you as I've studied the datasheet. As you've found `cmd5` should definitely be `False`.

peterhinch, 2022-06-24:
Just to clarify the business of SPI pins on RP2040. The [MicroPython docs](https://docs.micropython.org/en/latest/rp2/quickref.html) indicate that SPI0 has a few legal pinouts. Therefore I would assume that you'd need to specify which one to use:
```python
from machine import SPI
spi = SPI(0, 10_000_000, sck=2, mosi=3, miso=4)  # for these pins
```
Is this correct or is there an error in the official doc?

r00t1900, 2022-06-24:
I think this is correct in micropython, but not point out in rpi pico official docs. In its docs, pin group involved with GP17 are declared as the default SPI0 Pins, and other SPI0 Pins are the alternative/reconfigure Pins.## evidencepicture in pages of https://datasheets.raspberrypi.com/pico/pico-datasheet.pdf

peterhinch, 2022-06-25:
OK, thank you.