peterhinch/micropython_eeprom #20: micropython-I2C ony able to save up to 64 bytes size files on FT24C256A I2C EEPROM

Kyuchumimo, 2023-10-31:
Hardware: Raspberry Pi Pico (RP2040)
Firmware: Micropython v1.21.0
EEPROM: FT24C256A (32kB)

Example code:
```
import os
from machine import Pin, I2C
from eeprom_i2c import EEPROM, T24C256
eep = EEPROM(I2C(1,sda=Pin(2), scl=Pin(3), freq=400000), T24C256)
# Format the filesystem
os.VfsLfs2.mkfs(eep)  # Omit this to mount an existing filesystem
os.mount(eep,'/eeprom')

with open('/eeprom/main.py', 'w') as file:
    file.write('from machine import Pin, Timer\nled = Pin(25, Pin.OUT)\ntimer = Ti')
with open('/eeprom/main.py', 'r') as file:
    print(file.read())
print(os.listdir('/eeprom'))
```

If I try to type one more character at file.write(), the system throws an `OSError: 28 ("No space left on device")` error.

peterhinch, 2023-11-01:
Have you run the [tests](https://github.com/peterhinch/micropython_eeprom/blob/master/eeprom/i2c/I2C.md#5-test-program-eep_i2cpy)?

Kyuchumimo, 2023-11-01:
**test()**:
```
1 chips detected. Total EEPROM size 32768bytes.
Instantiated EEPROM
Test of byte addressing passed
Test of slice readback passed
Test block boundary 256 passed
Test chip boundary skipped: only one chip!
```
**full_test(block_size=64)**:
```
Page 0 passed
Page 1 passed
...
Page 510 passed
Page 511 passed
```

**fstest(format=True)**:
```
1 chips detected. Total EEPROM size 32768bytes.
Instantiated EEPROM
Contents of "/": ['eeprom', 'bdevice.py', 'eep_i2c.py', 'eeprom_i2c.py']
Contents of "/eeprom": []
(512, 512, 64, 62, 62, 0, 0, 0, 0, 255)
```

**cptest()**:
```
1 chips detected. Total EEPROM size 32768bytes.
Instantiated EEPROM
Device already mounted.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 137, in cptest
  File "<stdin>", line 32, in cp
OSError: 28
```

peterhinch, 2023-11-02:
To date errors like this have resulted from electrical problems. Please check:
 1. The EEPROM is powered from 3V3. They are not 5V compatible.
 2. The I2C lines have external pullup resistors to 3V3. Value should be in the range 1KΩ to 5KΩ.
 3. I2C wires should be short.
 4. There should be a short, direct Gnd connection.

Kyuchumimo, 2023-11-02:
The memory I am using comes in the form of a module (on a board), which contains most of the components necessary to make it work.
In my case, the only thing this module lacks is the decoupling capacitor for power-up mentioned in the data sheet. Some modules have it, others do not. However, by placing a 22nf capacitor between VCC and GND, it is the same result in all tests.

1. FT24C256A is 2.5V-5.5V tolerant, but Raspberry Pi Pico is only 3.3V tolerant, so it is already considered.
2. Pull-up resistors are 10k ohm
3. I2C jumper wires are 3.25" inch long
4. No short-circuit at all

peterhinch, 2023-11-03:
I attempted to replicate your setup as closely as possible. I can't replicate the problem. Test conditions:
 * Pico W running release build 1.21.
 * 24AA512 I2C EEPROM (unfortunately my 256KiB chips are unavailable: soldered onto PCB test jigs).
 * 2.2KΩ pullups to 3V3 Pico output.
 * Device powered from 3V3.
 * eep_i2c.py adapted for I2C(1).
 * All tests pass including file copy test.

I'm not sure what to suggest. I was conscious that I hadn't tested with RP2 or firmware V1.21, but this test has ruled them out as the source of your problem.

I'm open to suggestions but at the moment I can't see what more I can do.

peterhinch, 2024-01-09:
There was a bug which was probably causing your problem. Please try the latest version.

Kyuchumimo, 2024-01-09:
I tried the cp() (CoPy) function to copy a file from the internal flash memory to the external eeprom memory and now it works correctly.

closed 2024-01-09 by Kyuchumimo

peterhinch, 2024-01-10:
Thanks for the report. Apologies for this nasty bug.