peterhinch/micropython_eeprom #25: TypeError: 'Pin' object isn't iterable @ flash_spi.py

Kyuchumimo, 2024-03-17:
The cspins parameter @ flash_spi.init assumes a tuple with multiple Pin objects to chip select pins for multiple Flash memories, but what if I only want to use a single Flash memory?
```py
38 ...        for cs in cspins:  # Deselect all chips
39 ...            cs(1)
```

This is my configuration:
```py
cspins = (Pin(5, Pin.OUT, value=1))
flash = FLASH(SPI(0, baudrate=20_000_000, sck=Pin(2), miso=Pin(4), mosi=Pin(3)), cspins)
```

Kyuchumimo, 2024-03-17:
Solved by using a list. The documentation (FLASH.md) should be changed from saying that cspins is a tuple to a list instead.

2. cspins A list or **tuple** of Pin instances. Each Pin must be initialised as an output (Pin.OUT) and with value=1 and be created by machine.

peterhinch, 2024-03-18:
I think you're missing a requirement of the Python language. While it is possible to define a list with a single element
```python
my_list = [42]
```
to do the same for a tuple the syntax is
```python
my_tuple = (42,)
```
The trailing comma enables the runtime to distinguish tuple creation from the use of parentheses in an arithmetic context.

closed 2024-03-18 by peterhinch