peterhinch/micropython_eeprom #26: Simple question : FM24CL64

patfrench, 2024-04-10:
Hello peter,

my question is perhaps stupid but I would like to know if my project is feasible:

I want to use my pyboard to :
1. load a file from my PC
2. program a 24CL64B which is on another PCB.

I don't want a filesystem because my pyboard will only be used as a programmer. I'm running Windows 1.

do you think this is feasible? 

thank you

closed 2024-04-10 by peterhinch

patfrench, 2024-04-10:
I want to use the pyboard but can I load the file from my memory through the REPL and then load the eeprom?

I just wanted to have your point of view with your experience

peterhinch, 2024-04-10:
OK, from your OP I couldn't see how MicroPython fitted in.

You could use a Pyboard as a programmer. The library enables a connected device to be viewed as an array of bytes, with no filesystem. On a PC you could create a binary file and transfer it to the Pyboard using a tool such as the official [mpremote](https://docs.micropython.org/en/latest/reference/mpremote.html). A very simple Python script would open the file and transfer its contents to the device.

If you read up on mpremote you will see that doing this would be a one-liner on the PC: mounting the PC filesystem onto the Pyboard and running the Pyboard's Python script (which would directly access the file on the PC).

If you need help with this I suggest you open a question in [discussions](https://github.com/orgs/micropython/discussions).